DYefremov/DemonEditor #151: Enigma crash on favorite bouquet after editing

jjg3000, 2023-01-24:
Hello,

On a Vu+ Duo2 with enigma2 VTI Image.

After a transfert from the STB, then assigning a Picon to a service and then sending back to the receiver my favorite list crash the STB and restart Enigma. THe service list is fine and I can scroll down the services but when the list should show the modified service it reboots the GUI..

Removing the service or completely rebooting the STB makes no change, my favorite list is unusable... 

Thanks anyway for what will become a great tool !

DYefremov, 2023-01-24:
Hi.

> assigning a Picon to a service and then sending back to the receiver my favorite list crash the STB and restart Enigma.

Assigning a Picon to a service does not change the list of favorites and has nothing to do with it! Perhaps the problem is something else. We can not exclude the problem of the image/hardware. You can send me a compressed "data" folder with your settings (to the email from my profile) indicating which channel is causing the problem for you. I can check on my box.

jjg3000, 2023-01-24:
[userbouquet.favourites.zip](https://github.com/DYefremov/DemonEditor/files/10488879/userbouquet.favourites.zip)
Thank you for your quick answer,
I don't know what exact service is causing the crash. Enigma crash when I scroll down when the last service in the list is Tv5 Monde. It can then be any of the 10 services that should be listed after Tv5 Monde. 
Thinking of it I also modified some of these 10 services with your tool without problem on the STB before changing also the Picons. I changed the Types from "Non-TS" to eServiceUri. 
Of course I removed the Picons but the problem is still there. I'm gona try to change back the types... 

Thank you again.

DYefremov, 2023-01-24:
> I changed the Types from "Non-TS" to eServiceUri.

It looks like this is the problem. Apparently your image does not support this type. Perhaps there is no additional plugin, etc.

closed 2023-01-24 by DYefremov