DYefremov/DemonEditor #154: epg - filter

luxorko, 2023-01-28:
hurried, selection by sattelite showing but actually deamon still make "search and parse" from whole (ALL) channels,
instead from just selected channels list.
could you look at this please

DYefremov, 2023-01-28:
Hi,
It's not entirely clear what you mean.

luxorko, 2023-01-28:
please look at issue #148 (now closed)

DYefremov, 2023-01-28:
Still, I don't quite understand what's wrong.:(
[image]

luxorko, 2023-01-28:
when choose "auto" demon should search only from list (etc Astra 19.2) ...instead from all satellites 
..he parsing epg from other satellite and not from filter

DYefremov, 2023-01-28:
Changed! There is still room for this feature improvement, but significant changes are required.

luxorko, 2023-02-11:
ok

if using XML file 

[image]

table are not completed; lost of details missing

[image]

of course "filters" not work ( when reading from xml file)

DYefremov, 2023-02-11:
> if using XML file

This is logical. We do not have a "position" parameter. Theoretically, you can use the value from "namespace", but I'm not sure if this makes any **practical** sense. Usually these files are not used for all satellite positions, but for certain ones.

luxorko, 2023-02-11:
Rytec source have lot (maybe ALL) of satellite referencies for Europe
`http://www.xmltvepg.nl/rytec.channels.xml-sat.xz` 
its looks like:

`<channel id="NASAUHD.us">1:0:1F:68:C:A:E080000:0:0:0:</channel>`

if demon will filtering xml list  - showing them by sat/channelID/Channel - will be great option  (meaning easy to do) for parsing epg

just imagine: you have list m3u (somewhere from net) where are more channes (100+) etc you like Portugals  
1 - rytec provided - sat referencies + epg 
2 -demon will parse iptv bouquet with rytec reference (downloaded from url) or any ITV provider reference list
3 - stb enigma2 will download epg throught epgimport 
.Finally = IPTV Bouquet with EPG!

sound is good ; OR?

DYefremov, 2023-02-11:
OK. I don’t see any big problems with the implementation of displaying satellite positions for the XML source. But... 
Not all files have this:
>  -- 0.8W --

Therefore, the main option, as I wrote above, is the "namespace". Anyway, I'll see what I can do. :)

luxorko, 2023-02-11:
maybe by "channel ID"   (nassauhd.us ; 3sat.de; tfx.fr...) 
one channel could be on (3-5) sat. possition - so i thing filter by ID will be better option maybe?
3sat.de EPG must be same for all sattelite possition in think

luxorko, 2023-02-11:
also iptv provider xml-list have.. ID Channel  ...reference....channel name
some iptv provider using "Channel ID" like number (257) , but third column have actual channel name where will by parsing that reference.

DYefremov, 2023-02-11:
Unfortunately there is no standard here. Everyone can invent their own bicycle and write as they please.:) While we need to stop at the position, and then we need to look.

luxorko, 2023-02-12:
agree with you...please make vissiblle FULL ID Channel name  - includind punctuations marks  ( . - / _ )
"NASAUHD.us" and not just "NASAUHD"

DYefremov, 2023-02-12:
> please make vissiblle FULL ID Channel name

You should not confuse the channel ID with its name. Anything can be used as an ID. E.g. number, reference, etc. The fact that the ID almost matches the channel's display name is just a special case! 
[image]

Perhaps, if an XML source is used, it is better to display a separate column with id... Сurrently the 'id' attribute is ignored as unnecessary!

luxorko, 2023-02-12:
I think this is the best option.

100% .. im with you

luxorko, 2023-02-16:
great ...waiting for new beta realease

luxorko, 2023-02-17:
update 3.4.0
1 - filter by sat. possition workin - great
2 - seaching service by name working but- missing full channel name (including ".de" or ".hr")...eurosport.de ; eurosport.hr...

[image]

3 - parsing Reference from .xml ro iptv channel not working :(
all channels from .xml file will always use same same reference number (

4097_0_1_1_1_1_D7A312D_0_0_0`

luxorko, 2023-02-17:
from "xml ID" file.....have same channel like channels Eurosport  diferent EPG for 3 country please see pictur

[image]

DYefremov, 2023-02-17:
> missing full channel name (including ".de" or ".hr")

**This is not a full name!** Please read my [previous answer](https://github.com/DYefremov/DemonEditor/issues/154#issuecomment-1427008437). The channel name is the one on the right side of your screenshot.

> have same channel like channels Eurosport diferent EPG for 3 country please see picture

Sorry, but this question is not for me, but for the source.

> 3 - parsing Reference from .xml ro iptv channel not working :(
> all channels from .xml file will always use same same reference number

Please provide a link to the file, I'll check.

luxorko, 2023-02-17:
please see link

`http://www.xmltvepg.nl/rytec.channels.xml.xz`

and not possible to get both side channels name+channel ID  info to vissible in demon? pleasde

DYefremov, 2023-02-17:
> get both side channels name+channel ID info to vissible in demon

Please read my [answer](https://github.com/DYefremov/DemonEditor/issues/154#issuecomment-1427008437). 

> Perhaps, if an XML source is used, it is better to display a separate column with id

> I think as an option to display a tooltip.

I made a tooltip:
[image]

> http://www.xmltvepg.nl/rytec.channels.xml.xz

Tested on several channels. You can specify which channels can not assigned?

luxorko, 2023-02-17:
open any IPTV bouquet.tv
then parsing first one nasauhd to any iptv channel
then save and open saving bouquet list and that channels will have THAT reference number

luxorko, 2023-02-17:
I made a tooltip: 
super !

luxorko, 2023-02-17:
please see pictures
before parsing
[image]

 parsing
[image]

and after parsing
[image]

DYefremov, 2023-02-17:
> then parsing first one nasauhd to any iptv channel

Hm. In this version, I added a channel type assignment (as for the main list). The problem seems to be this (**1F** value, UHD channels).

luxorko, 2023-02-17:
it only exapless ...this happens with any reference from xml file parsing to any iptv channel

also if IPTV channel have basic reference 4097:0:1:0:0:0:0:0:0:0  still same result like you see on last picture No3

luxorko, 2023-02-17:
"1F" - meaning channel is UHD : ) i think you know it

DYefremov, 2023-02-17:
Found. The reason isn't the type. It was tooltip. He "broke" everything, or rather my inattention. :)

luxorko, 2023-02-17:
hope you will kept "tooltip" is very usefull

luxorko, 2023-02-17:
could you new realease beta for more test, please

DYefremov, 2023-02-17:
As soon as I have free time. Maybe tomorrow or later. While I was looking for a problem with refs assigning, I noticed one more nuance in one other feature. Need to check.

luxorko, 2023-02-17:
ok dont worry

regarding "filter-searching" channel by name 

could you do something with ..some IPTV channel have in the front channel name "BG" and also behind name (TV; HD; FHD; 1080...) like on picture

idea when start searching Channel name from IPTV " [BG] DIEMA TV HD" - demon not find any, need manually update name to "DIEMA" and then will showing lots channels

please see picture. hope you know what i meaning

[image]

DYefremov, 2023-02-17:
> demon not find any, need manually update name to "DIEMA"

What is the criterion? What symbols? And why exactly these? What separator to use? And most importantly. Filtration speed after that. No it's a bad idea.

Perhaps you need a filter like the filter on the picons tab. you are looking for several words at once through the symbol '|'. Or add channel id to filtering:
[image]

 BUT again it doesn't solve what you want.

luxorko, 2023-02-26:
have idea - why not make turn /switch searching table? search from source to m3u list?
Could demon have that kind of switch?

DYefremov, 2023-02-26:
Sorry... Describe in more detail.

luxorko, 2023-02-26:
EPG configuration
have two tables
left EPG Source ; and right bouquet (in our case IPTV-bouquet) like on picture
[image]
on Right table (iptv bouquet Channels have name Bollywood classic (EU) )
- activate "AUTO"
[image]

 click on channels name "bollywood classic (EU) = nothing show onb left table (EPG Source)
you said :
`

    demon not find any, need manually update name to "DIEMA"

What is the criterion? What symbols? And why exactly these? What separator to use? And most importantly. Filtration speed after that. No it's a bad idea.`

idea make just "SWICH" (where get search from); not from IPTV Bouquet (right table) but from EPG source (left table)
 
using filter; choose sat. possition 
[image]

and there click (epg source list) channel name "Bollywood" and on the right table (IPTV Bouquet) will showing all "Bollywood" channels including all symbols "HD; (EU), PT, FHD...  and what ever this channel name have it) 
after just making parse EPG (parse  references) like this function have now

hope you know what i mean

DYefremov, 2023-02-26:
Like understood. You want change the filter field on click not  from the bouquet only but  from the source (**if auto is enabled**) list also. I don't see any problems. I think, no additional switches is needed.  We  have an "Auto" switch already.

luxorko, 2023-02-26:
` Možno to bude pre používateľov jasnejšie`

ok like you said

luxorko, 2023-02-26:
`Perhaps it will be more obvious to users`

ok like you said

DYefremov, 2023-03-04:
It was some time to experiment...
[image]
I think it's completely useless and inconvenient. Each time you need to select a new channel, you need to clear the input field. At the moment I see no reason to add anything else here! Coming up with additional logic for clearing a field is just confusing the user.

DYefremov, 2024-07-18:
After various variations, I came to the conclusion that the option described above would be more confusing for the user. Since after filtering in the first window, it may be necessary to make a new selection by name. In general, I think it should be left as is.
Perhaps in the future, if the feature is really in demand, it will be possible to think through other options. At the moment we're closing...

closed 2024-07-18 by DYefremov