DYefremov/DemonEditor #157: picon location

luxorko, 2023-02-16:
cannot manually change location for transfer picons
writing manually - and demon will adding endline these "\" see picture

could you add to locatian for picon please
/usr/share/enigma2/picon 
/hdd/picon
/usb/picon

[image]

DYefremov, 2023-02-16:
Hi,

> /usr/share/enigma2/picon

This path is present by default. 
[image]
Perhaps you removed it. After resetting the settings should appear again.

> /hdd/picon
> /usb/picon

Added to defaults. It should also appear after settings reset.

> demon will adding endline these `"\"`

I think it's related to Windows build only. Corrected.

luxorko, 2023-02-16:
ok 
maybe have option "blank" do not import picon at all?
yes /no ?

DYefremov, 2023-02-16:
> option "blank" do not import picon at all?

Hmm, I don't quite understand what you mean.

luxorko, 2023-02-16:
you can choose where should be transfer picon to receiver
but of user dont wanna have picons becouse old receiver are slow?

he can see picons in demon but make "block-do not transfering" any to reciver/hdd/usb ...

DYefremov, 2023-02-16:
> he can see picons in demon but make "block-do not transfering" any to reciver/hdd/usb ...

And what forces you to load picons into the receiver now?  Just don't download.  To display in the program, you do not need to have piсons in the receiver at all. I have several thousand picons on my computer (to display im DemonEditor), and in the receiver only those that are in my favorite lists (about 70). You can download the desired picons from the Web or add from a local folder:
[image]
You can download only the necessary piсons by simply selecting the ones you need from the list and clicking the "Download" button.

closed 2023-02-26 by DYefremov