DYefremov/DemonEditor #164: iptv stream

luxorko, 2023-03-07:
demon v 3.5.0 have bugs

1-Downloading bouquets from stb not closed at the end
[image]

2 working stream detected as not working ( not playing) and when checking-removing

`http://84.244.187.150:1935/live/smil:myStream.smil/chunklist.m3u8
`
[image]

DYefremov, 2023-03-08:
> demon v 3.5.0 have bugs

There no changes in work with data download and streams from previous version! 

> 1-Downloading bouquets from stb not closed at the end

Have you already checked the download on the previous program version? I can't reproduce it. Perhaps you have changed your bouquets in third-party plugins or manually.

> working stream detected as not working ( not playing) and when checking-removing

Can't confirm deletion. At me it's OK. In terms of playback. This problem exists, but only on **Windows**. Perhaps the problem is in one of the libraries. For example,  don't 'likes' link format (or some symbol).

luxorko, 2023-03-08:
regarding downloding:

v 3.4.2 - no problem

[image]

v3.5.0 - have problem

[image]

luxorko, 2023-03-08:
`Can't confirm deletion. At me it's OK. In terms of playback. This problem exists, but only on Windows. Perhaps the problem is in one of the libraries. For example, don't 'likes' link format (or some symbol).`

These URL (stream) are playing on VLC and PotPlayer without any problem.., 

why demon cannot opeing stream on Pot player

[image]

or VLC

[image]

luxorko, 2023-03-08:
..and what wrong with these stream?  also not works on demon

`https://tv-trtkurdi.medya.trt.com.tr/master.m3u8`

bouquet looks 

`#SERVICE 4097:0:1:1:0:0:0:0:0:0:https%3a//tv-trtkurdi.medya.trt.com.tr/master.m3u8
#DESCRIPTION TRT Kurdi`

luxorko, 2023-03-08:
looks like demon have problem playing streams with end ".m3u8"

DYefremov, 2023-03-08:
> v3.5.0 - have problem

It's strange...  I checked with all my presets. In the latest version, I added an additional optimization that does not allow you to load duplicates to the main list.  Perhaps this is related. It is advisable to get your full set of settings to check. Just zip your settings folder (all bouquets and lamedb) and send me on email for my profile. 

> and what wrong with these stream? anot working

This is on Windows:
[image]

> why demon cannot opeing stream on Pot player

Since this player has nothing to do with this program. The Windows build uses [MPV](https://mpv.io/) by default. The library is already included in the package.  You can optionally use VLC. But correct operation on WIndows is not guaranteed. VLC should be installed by default (int to Program Files).

> These URL (stream) are playing on VLC

The problem may be in other libraries as well. Also, blocking stream playback on the server side cannot be ruled out (GEO blocking etc). This issue is secondary. The main thing is that with the **data loading**. I repeat. All my presets load without problems.

luxorko, 2023-03-08:
found it

problem are in bouquet.tv  (IPTV Stream)

`#SERVICE 4097:0:1:1:0:0:0:0:0:0:https%3a//edge-uk2.manoto.click/live.m3u8

#DESCRIPTION Manoto TV
`
please see there are missing ":Manoto TV" behind m3u8 at end 
once adding and stream playing OK

DYefremov, 2023-03-08:
> problem are in bouquet.tv (IPTV Stream)
> 
> `#SERVICE 4097:0:1:1:0:0:0:0:0:0:https%3a//edge-uk2.manoto.click/live.m3u8
> 
> #DESCRIPTION Manoto TV

And how did this stream end up in this file?

luxorko, 2023-03-08:
regarding downloading bouquet from stb to demon:

-open demon (do not click else just on icon for downoading)
-click straight on downloand icon
[image]

and error showing

if 
-open demon
- click anywhere like on "File or bouquet list...."
- then downloading icon = error not showing 

could you repeat please

luxorko, 2023-03-08:
> And how did this stream end up in this file?

enigma2 stb streaming correctly, if adding channels name behind m3u8 also no problem

DYefremov, 2023-03-08:
> enigma2 stb streaming correctly, if adding channels name behind m3u8 also no problem

This file is not for that. If you add something manually, you must know exactly what you are doing.

> could you repeat please

I don't see any problems.

DYefremov, 2023-03-08:
> > enigma2 stb streaming correctly, if adding channels name behind m3u8 also no problem
> 
> This file is not for that. If you add something manually, you must know exactly what you are doing.
> 
> > could you repeat please
> 
> I don't see any problems.

P.S. If such problems occur, without the full set of settings, such requests will be closed and ignored!

luxorko, 2023-03-08:
> 

> P.S. If such problems occur, without the full set of settings, such requests will be closed and ignored!

this what i done now; also checked last 5 version of Demon and same behaviors

luxorko, 2023-03-08:
> This file is not for that. If you add something manually, you must know exactly what you are doing.

now know what causes a problem. (Incrorrect created bouquet) .bouquet got from net. somewhere

should be closed now.

DYefremov, 2023-03-08:
Ok. 

> enigma2 stb streaming correctly, if adding channels name behind m3u8 also no problem

What works for you doesn't mean it will work for everyone! Enigma2 is a big zoo. And I have neither the time nor the desire to guess what and how you manually enter into your files.  Sorry. But...

Closed!

closed 2023-03-08 by DYefremov

luxorko, 2023-03-08:
its realy strange 
this ara url stream

`https://edge-uk2.manoto.click/live.m3u8`

if thease url add to Demon like new IPTV channel  - its working
and bouquet looks like 
`#SERVICE 4097:0:1:0:0:0:0:0:0:0:https%3A//edge-uk2.manoto.click/live.m3u8:manot

#DESCRIPTION manot`

if i add bouquet.tv  without  "m3u8:manot" to demon - stream not working in Demon but working in enigma2 receiver 

`#SERVICE 4097:0:1:0:0:0:0:0:0:0:https%3A//edge-uk2.manoto.click/live.m3u8

#DESCRIPTION manot`
 
any idea ??

DYefremov, 2023-03-08:
> any idea ??

I think if you use an editor, then you need to use an editor. If you want to enter data manually, then I see no point in using the editor.  DemonEditor uses a **valid type of IPTV description** (format) - and this is the main thing. And it also uses this format to get information about the link. Enigma2 (all or some distributions) may take a different approach, ignoring or omitting any data (eg stream description).