DYefremov/DemonEditor #165: Merge multiple transponder lists in a single orbital position ?

lgrawet, 2023-03-10:
Hi,

When updating transponder lists from Kingofsat (or LyngSat...), is it possible to merge transponder lists on the same orbital position ?
For example, Hot Bird will be splitted in three transponder lists and the preexisting 13E of the file will be left untouched. (open)enigma will then complain about invalid satellites.xml file

13.0°E 	Hot Bird 13B
13.0°E 	Hot Bird 13E
13.0°E 	Hot Bird 13C

Thank you,

Laurent

DYefremov, 2023-03-11:
Hi,
I think it's possible. E.g as an additional option:
[image]

DYefremov, 2023-04-06:
Added base implementation. Not fully tested!

closed 2023-04-15 by DYefremov