DYefremov/DemonEditor #167: demoneditor on termux-xfce4 on android-aarch64

risky777, 2023-03-12:
Hi,

I wanted to see if demon-editor will run on my 10" android tablet with arm64-cpu on termux-xfce4. I managed to get it working with just few more steps than on standard linux, and all additional steps are reasonable except one.

When I run "python start.py" I get error:

File "/storage/emulated/0/Download/demoneditor/app/ui/uicommons.py", line 85, in <module>
locale.bindtextdomain(TEXT_DOMAIN, LANG_PATH)
OSError: [Errno 22] Invalid argument

I bypassed that error editing this file (uicommons.py) and commenting-out lines 83-84
#if UI_RESOURCES_PATH == BASE_PATH:
	#locale.bindtextdomain(TEXT_DOMAIN, LANG_PATH)
	
Do you know what could cause that?

DYefremov, 2023-03-12:
> Do you know what could cause that?

This code is used for localization. In your case it looks like the method signature is different for the platform you're using. I can't say more exactly. You need to look at the module itself.

closed 2023-03-13 by DYefremov

risky777, 2023-03-15:
Just for info (if someone else tries on termux):

 https://github.com/termux/termux-packages/issues/15626

Try patch would be nice, but it's not important, important is that it works.