DYefremov/DemonEditor #168: DemonEditor-3.5.0-Beta.deb gives "550 failed to change directory"

hugocoolens, 2023-03-13:
I installed DemonEditor-3.5.0-Beta.deb on Linux Mint 21. I entered the ip adres, port, password... of my receiver and did a connection test which was successful. Unfortunately when trying to fetch the data from my receiver to the computer I always get the
error "550 failed to change directory". Is this a bug or am I doing something wrong?

DYefremov, 2023-03-13:
Problem with FTP access to files on your Box. 
1. Check that the Enigma2 paths are correct:
[image]
2. For all recent Enigma2 images, it is mandatory to **set a password** for "root" user. Without this, there will be no access also.
On your Box : System->Network->Password Setup

For more information, look on the support forums for your image.

hugocoolens, 2023-03-15:
> 

Thanks this solved the issue

closed 2023-03-15 by hugocoolens

hugocoolens, 2023-03-15:
thanks, problem solved

kind regards,
Hugo
p.s. is it possible to send a reload command to the receiver?

DYefremov, 2023-03-15:
If the HTTP API is enabled, then this request is called automatically.
[image]

hugocoolens, 2023-03-15:
OK, thanks!

kind regards,
Hugo