DYefremov/DemonEditor #169: channels show <n/a> and channel names replaced by  "DVB service"

hugocoolens, 2023-03-17:
After sending my modified channel list to my receiver, a lot of 
channels show <n/a> in stead of their name on the receiver, the channels 
do play however and in the bouquet list of demon editor the names are 
visible. I also noticed that those channels show the words "DVB service" 
in stead of the channel name on my TV.

DYefremov, 2023-03-17:
Maybe you didn't do something.

> After sending my modified channel list to my receiver

Which specific list did you modify? Main? It's not entirely clear what and how you did. 

> channels show <n/a> in stead of their name on the receiver

If this is displayed in your bouquet, then usually this means that these services are not in the main list (they are not in lamedb).
But looking by this:
> in the bouquet list of demon editor the names are  visible

 everything should be ОК.

A similar thing can also happen if you have **incorrectly configured** satellites (LNB) of your receiver.

P.S. You can always restore previous settings via "Tools -> Backups". But I think that the problem is clearly not in the editor and settings.

hugocoolens, 2023-03-17:
> Maybe you didn't do something.
>
> After sending my modified channel list to my receiver
>
> Which specific list did you modify? Main? It's not entirely clear what and
> how you did.
I choose a Satellite and added a new transponder, sent it to my receiver,
did a scan on my receiver which worked fine, then added some new found
channels to my bouquets on the receiver. Fina

hugocoolens, 2023-03-17:
> Maybe you didn't do something.
>
> After sending my modified channel list to my receiver
>
> Which specific list did you modify? Main? It's not entirely clear what and
> how you did.
I choose a Satellite in demon-editor and added a new transponder, sent it
to my receiver, did a scan on my receiver for that new entry which worked
fine, then added some new found channels to my bouquets on the receiver.
Did a transfer from
receiver to computer and vice versa. Thereafter all those n/a and DVB
service entries became visible in the bouquets on my receiver.

> channels show <n/a> in stead of their name on the receiver
>
> If this is displayed in your bouquet, then usually this means that these
> services are not in the main list (they are not in lamedb).
> But looking by this:
>
> in the bouquet list of demon editor the names are visible
>
> everything should be ОК.
It should be, but it is not unfortunately

> A similar thing can also happen if you have *incorrectly configured*
> satellites (LNB) of your receiver.
I don't think so, as I have installed dreamboxedit with crossover once
again and everything is fine again.

> P.S. You can always restore previous settings via "Tools -> Backups". But
> I think that the problem is clearly not in the editor and settings.
thanks for the suggestion
kind regards,
Hugo

DYefremov, 2023-03-17:
> I choose a Satellite and added a new transponder, sent it to my receiver,

Perhaps the name (parameters) of the satellite differ from the previously configured one.

> I don't think so, as I have installed dreamboxedit with crossover once again and everything is fine again.

Perhaps a simple reload of the settings helped.

In any case, the question is about the Enigma2 settings - configuration (reception) itself (which glitches should also not be ruled out.).

DYefremov, 2023-03-17:
> I don't think so

I had a similar experience, so I'm not just writing. :)

closed 2023-03-18 by DYefremov