DYefremov/DemonEditor #170: Boquets entry with Youtube URL- detection and formating.

regrepl, 2023-03-22:
At the beginning I will say thanks for developing this program, it looks promising, but I discover some errors that I want to share with you.

One problem is about detecting entries with youtube URL and second is about formatting entries with youtube URL.

- First problem is when I try to edit IPTV service that contains youtube URL.

When I want to edit this entry I get a popup window. I'm clicking cancel, and then another popup which is exactly the same, second time cancel and it is finally able to edit.
Moreover when I'm pasting new url immediately the same popup appears or even when I want to modify the reference the same reaction. That is pretty annoying error.
[image]
The worst part is that there is no option to disable this detection. Please add that option.

I also tried to click ok but then, I got a info about conflicted versions with system one (linux), after solving that there is another problem.
A little information ahead. To watch youtube directly from channel list I need to have additional plugins from feed installed that act like proxy either is is streamlink or warpers like yp-dlp (introduced in for example openatv 7.x)
So the url is formatted with header it can be **"http://127.0.0.1:8088/"** or **"YT-DLP://"** but that header cannot be transferred to youtube-dl because it is causing problems, without header clicking ok is fine, but that is not the point to have an entry that box will not process.

- Second problem has to do with the formatting of this type of URL entry, unfortunately there is an additional conversion to ASCII where it should not be converted. And this is more serious.

Proper URL
`YT-DLP://https://www.youtube.com/watch?v=i7D8laFtDsw`
Proper bouquet entry
```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:YT-DLP%3a//https%3a//www.youtube.com/watch?v=i7D8laFtDsw:YT CAM 1
#DESCRIPTION YT CAM 1
```

Entry after modification with DemonEditor, so I will call it corrupted bouquet entry
```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:YT-DLP%3A//https%3A//www.youtube.com/watch%3Fv%3Di7D8laFtDsw:YT CAM 1
#DESCRIPTION YT CAM 1
```
Where **? and =** got converted as %3F and %3D, that should not be converted only **":"**
I never saw that behavior in any other program, but formatting url like that may have influence on other IPTV  links not only YouTube.

And of course this is not working in the box due to bad syntax

Different method and this time I will present only proper entry
```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:http%3a//127.0.0.1%3a8088/https%3a//www.youtube.com/watch?v=i7D8laFtDsw:YT CAM 1
#DESCRIPTION YT CAM 1
```
I tested this on linux version 3.5.0 Beta

I hope that reproducing and fixing that error will not be a big problem.

DYefremov, 2023-03-23:
Hi,
First of all, there are no errors here. Everything works as intended (Auto detection of direct links to YouTube). Yes, it might make sense to optionally disable links auto-convert. Maybe I'll even take YouTube-related features into a [separate extension](https://github.com/DYefremov/demoneditor-extensions).Or I'll add an option to the IPTV dialog itself.  I'll think.

regrepl, 2023-03-23:
Work as intended, then in current state I don’t see any practical application for that function.
Separate plugin seems to be a good option, that allows to do more than additional checkbox, however both options have positives.

I tested also Windows version it is even worse. I made a video, that may better elaborate what I’m consider as error.

https://user-images.githubusercontent.com/39137187/227374736-bed3affc-c328-4c66-a8ab-555d00d95191.mp4

It  reminds me those old fashion way web advertisement when you had to chase X to close the window. When dialog box requires user to additional reaction and after that reaction you see the same dialog box in different position, you can start suspecting a false loop. Fortunately it ends at second time. 
In Linux version popup position and edit window is more persistent and it always the center of app, but also popup appears two times before allowing to edit.

DYefremov, 2023-03-24:
> I don’t see any practical application for that function.
 
But I see... :) I don't need additional plugins. I don't have to worry about inserting various prefixes and etc. to briefly look at any video or playlist. Аnd there are enough such users.

> popup appears two times before allowing to edit.

Yes, this looks like a bug. I'll take a look.

DYefremov, 2023-03-25:
> popup appears two times before

Should be fixed.

Regarding this:
> Proper URL
> `YT-DLP://https://www.youtube.com/watch?v=i7D8laFtDsw`

and this:
> I will present only proper entry

You should read about the "correct" URL and about escaping for it.  In your case (IMHO), this is a "**crutch**" for a particular plugin. 
Perhaps we should really add a separate option for post(pre)-processing such (xtreamity, etc.) links. As an option a drop-down list with the plugin name to select formatting. E.g, as made for the "Type" parameter. Here needs to think.

DYefremov, 2023-03-26:
Made a basic implementation.Not tested!
[image]

I hope it will work.

regrepl, 2023-04-01:
Thank you for your work and being open for suggestions. You are working very fast. Double popup is gone, that is very good news.

I used word correct in terms what enigma2 will recognize as correct entry and play it. So that's conditional based on environment.
The development of enigma2 is going toward supporting more iptv, so that’s means more plugins that can handle various platforms with links variables and tokens. I think that focus changed when images migrated to new version of gstreamer, after years of keeping 0.10 version or earlier when serviceapp was introduced. Actually wrapper were attempt to simplified some things. But that another discussion and doesn't change the fact that this solution started to be usage by images maintainers.

Back to the point, drop-down list is a great solution that I have not seen before, well though, easy to change and dose not interfere. That makes sense to me. I don’t have to wounder if prefix flex is correct, that simplifies usage of copy/paste function as well. However I didn’t manage to get playback, but that also stopped working me on older version, so may be something else.

This entry is not recognized as prefix. http://127.0.0.1:8088/. Internal url or prefix that may be debatable. This is actually for older solution called Streamlinksrv (not wrapper), default ip and port of course, that can be a changed. On older images more important than on newer ones. 
Nevertheless after using option edit and save, in bouquet file is ok with extra prefix, but in program is not displayed proper only “http://127.0.0.1” in url field. So perhaps a simple 4th option that is No prefix is best to add here.

Maybe off topic.

This type of drop-down list is a potential solution to other problems. Custom prefix option, or prefix as a variable defined in a separate option. Or something more complicated like a custom prefix list.

Perhaps this could also be a way to handle other solutions that use a prefix like Stream Relay, where the default port is usually used (17999), but can be easily changed. However, this solution would have to include all iptv services, not just those with youtube detected. So it is a bit of a departure from my original ticket. Thinking about how many cases enigma2 uses prefixes in url these days, and how it was in the past, makes me wonder how many prefixes will be in the future. Keeping track of what is new and what is old is also a complication. So maybe a prefix list as a future-proof feature is something to think about.

DYefremov, 2023-04-02:
> So perhaps a simple 4th option that is No prefix is best to add here.

Added. Not fully tested!
[image]

> This type of drop-down list is a potential solution to other problems. Custom prefix option, or prefix as a variable defined in a separate option. Or something more complicated like a custom prefix list.

I don't see a big problem. We can do, for example, as for the picons paths list in the settings dialog. But I think later. While there are other tasks.

regrepl, 2023-04-04:
One small thing after more tests. Case sensivity.

streamlink://
insted of
STREAMLINK://

Both program and wrapper threat this as different values, lowercase letters are correct.

Example of working entry
```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:streamlink%3A//https%3A//www.youtube.com/watch?v=ydYDqZQpim8:Nambia
#DESCRIPTION Nambia
```
Prefix was not recognized, anyway I tried to save and see what will happened, program added second prefix and the result in file was
```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:STREAMLINK%3A//streamlink%3A//https://www.youtube.com/watch?v=ydYDqZQpim8:Nambia
#DESCRIPTION Nambia
```
And then program was not able to edit it again. “:” is treated as end of url,
[image]

I tried to recreate it by editing boutique directly and adding only second preffix and got

```
#SERVICE 4097:0:1:0:0:0:0:0:0:0:STREAMLINK%3A//streamlink%3A//https%3A//www.youtube.com/watch?v=ydYDqZQpim8:Nambia
#DESCRIPTION Nambia
```

And this one could be edited and corrected inside program, ascii this time.
Some safety measure may need to be introduced to avoids accidents.

DYefremov, 2023-04-04:
Оk. Changed.

> lowercase letters are correct

Strange. Personally, I've seen examples of working links with an uppercase prefix. But in a zoo like Enigma2... :) I won’t be surprised if comes a request to reverse or change to uppercase with the new plugin version or next update of one of the Enigma images.

> Some safety measure may need to be introduced to avoids accidents.

Unfortunately, it is a very difficult task to foresee all the options for incorrect user actions.I don't think it makes much sense also.

regrepl, 2023-04-05:
> Unfortunately, it is a very difficult task to foresee all the options for incorrect user actions.I don't think it makes much sense also.

True, is hard to predict every situation like that.
`Invalid prefix for the given URL!` 
This is a clear info now and that disallows to create double prefixes.
Program also converts now uppercase entry to lowercase after edit (for streamlink), that is helpful,

I found one more thing during tests.
When prefix is used on non youtube entry. Program also detects that there is a prefix and doesn't show that part in edit window, only url is shown. After saving prefix is gone from bouquet file. At this stage non youtube entries are not the target and dose not have drop-down list, but are affected either way and are not filtered out.

DYefremov, 2023-04-06:
> I found one more thing during tests.
> When prefix is used on non youtube entry.

Yes, I missed this option. Thx. I have added additional check. I hope together we will bring this feature to a working state. :) Unfortunately it's difficult for one to notice all the nuances.

DYefremov, 2023-04-06:
> I found one more thing during tests.
> When prefix is used on non youtube entry.

Yes, I missed this option. Unfortunately it's difficult to notice all the nuances. Thx. I have added additional check. I hope together we will bring this feature to a working state. :)

regrepl, 2023-04-11:
I will get back shortly to case sensitivity. I was talking only about one prefix and tested that, didn’t notice you changed all, two other were correct before. Anyway I use a fragment of code to be clear now.

`_URL_PREFIXES = {"YT-DLP": "YT-DLP://", "YT-DL": "YT-DL://", "STREAMLINK": "streamlink://", "No": None}`
Upper, Upper, Lower, No

[added additional yt link checking (](https://github.com/DYefremov/DemonEditor/commit/7b002b208f07f1d6ab1003bd8a3871b9eae03bf0)[#170](https://github.com/DYefremov/DemonEditor/issues/170)[)](https://github.com/DYefremov/DemonEditor/commit/7b002b208f07f1d6ab1003bd8a3871b9eae03bf0) 

Yes, this change works very well.
I think that there is not much left to do.
Creating new IPTV Channel with YT link also works, nice feature that allows to get YT Title as channel name. That simplifies creation process.

There is also one special case when youtube links are used.
For live transmissions link to youtube channel ended with “/live”
Youtube_dl suports that as well, so also playback in program could be possible.

Example
`https://www.youtube.com/@kenyacitizentv/live`
or this described in youtube_dl code
`https://www.youtube.com/channel/UCoMdktPbSTixAyNGwb-UYkQ/live`
`https://www.youtube.com/@SkyNews`
`https://www.youtube.com/SkyNews`

This work best when there is one transmission per channel, so every start/stop of transmission means that “watch” link doesn't have to be updated. Although is not perfect, when there are multiple live transmissions on channel only one is selected, often not the main live broadcast on tv channels. With those new filter, when there is no yt link detection in this case, this can be defined as well regular with writing prefix manually. I don’t know if it is better to leave it that way or implement this exception, especially because use case is so limited. I mentioning it to be thorough. 

I also have one unrelated observation.
Function "Use HTTP to reload data in the receiver" is using only one webif command after sending channels. servicelistreload?mode=0 to be exact, is it possible to add another command servicelistreload?mode=4 ?
The second one allows to reload currently used bouquet in real time without the need to leave it to refresh it, so less box remote operations. Time saving, nothing big.

DYefremov, 2023-04-12:
> I was talking only about one prefix

Ok. Reverted.

> mode=4

I don't remember why, but I refused this option. It can be done optionally, but this requires a separate discussion.

As for streams playback...
This is a more wide task. For example how to (not) play links with prefixes (and not for YouTube links  only). I think this requires separate discussion also.

regrepl, 2023-04-13:
> Ok. Reverted.
Now is perfect.

> As for streams playback... This is a more wide task. For example how to (not) play links with prefixes (and not for YouTube links only). I think this requires separate discussion also.
I was thinking of something simpler, which is extending the youtube detection parameters by deleting the "watch?v=" part of the pattern, playback should be ok then. But I get it may be a part of fail safe about unforeseen actions or may be important to pass arguments to youtube_dl. Anyway, as I wrote, it was a special case, so not that important at this time.

With this comment I close the topic as all initial issues have been resolved or explained. As a result, we also have a new feature.