DYefremov/DemonEditor #171: [Help] Install on ChromeOS

giopas, 2023-03-27:
Hi,

I wanted to install DemonEditor on ChromeOs (v. 112), but I guess I need your help.

I have enabled "Linux development environment" and downloaded the .deb package from here and then typed the following lines on the terminal:

1. $ sudo dpkg -i DemonEditor-3.5.0-Beta.deb 
2. $ sudo apt-get install python3-requests python3-gi python3-gi-cairo gir1.2-notify-0.7 p7zip-full
3. $ sudo apt --fix-broken install
4. $ sudo dpkg -i DemonEditor-3.5.0-Beta.deb

I can now see DemonEditor on the App menu, but it does not launch. If I run it on terminal I get the following error:

`$ demon-editor 
Traceback (most recent call last):
  File "/usr/share/demoneditor/start.py", line 2, in <module>
    from app.ui.main import start_app
  File "/usr/share/demoneditor/app/ui/main.py", line 44, in <module>
    from app.eparser import get_blacklist, write_blacklist, write_bouquet
  File "/usr/share/demoneditor/app/eparser/__init__.py", line 33, in <module>
    from .enigma.lamedb import get_services as get_enigma_services, write_services as write_enigma_services
  File "/usr/share/demoneditor/app/eparser/enigma/lamedb.py", line 33, in <module>
    from app.ui.uicommons import CODED_ICON, LOCKED_ICON, HIDE_ICON
  File "/usr/share/demoneditor/app/ui/uicommons.py", line 36, in <module>
    gi.require_version("Gtk", "3.0")
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Gtk not available`

Is there something else I should do to make it work?
Has anyone succeded to install it/use it on ChromeOs?

Thanks

DYefremov, 2023-03-27:
Hi,
I am not familiar with this OS. But... I looks like you do not have the Gtk3 library installed.
For example, for Ubuntu/Debian these might be packages: libgtk-3-0, libgtk-3-bin, libgtk-3-common, libgtk-3-dev  etc.

giopas, 2023-03-27:
Thank you for the quick reply!

They all seems installed:

```
$ sudo apt-get install libgtk-3-0 libgtk-3-bin libgtk-3-common
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgtk-3-0 is already the newest version (3.24.24-4+deb11u2).
libgtk-3-0 set to manually installed.
libgtk-3-bin is already the newest version (3.24.24-4+deb11u2).
libgtk-3-bin set to manually installed.
libgtk-3-common is already the newest version (3.24.24-4+deb11u2).
libgtk-3-common set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
I have then looked a bit closer to the error "Namespace Gtk not available" and solved by installing those additional packages (not sure if all of them needed, but now it seems working:

`$ sudo apt-get install python3-gi gobject-introspection gir1.2-gtk-3.0`

Thank you!

I hope this thread will be useful for other users.

closed 2023-03-27 by giopas