DYefremov/DemonEditor #175: Error loading data from receiver (version 3.6.0 MacOS)

YoSafeguard, 2023-04-20:
Hi!

After updating the application to version 3.6.0 beta for MacOS Monterrey, when loading data from the receiver, the following error occurs:  **_LinksTransmitter: Load library error: Namespace AppIndicator3 not available_**

About a minute later, this is the entry: **_Downloading data error: [Errno 60] Operation timed out_**

Everything boots up fine on two MacOS Ventura computers
What does it mean? Thank you!

DYefremov, 2023-04-20:
Hi,
Unfortunately I can't reproduce it. I am using Formuler F1 and OpenATV 7.2.

> LinksTransmitter: Load library error: Namespace AppIndicator3 not available

It has nothing to do with data loading! It is playback bar warning.
[image]
In your case just **ignore it**!

> Downloading data error: [Errno 60] Operation timed out

It's more difficult. [Read this](https://github.com/jborean93/pypsexec/issues/5) first. Maybe your case. 
In any case, you need to check and test the FTP settings.
[image]
In addition, you can try to access your Box through the FTP tab or use a third-party FTP client (e.g. FileZilla).
I recommend [reading this](https://github.com/DYefremov/DemonEditor/issues/168#issuecomment-1466516470) also. You may have updated the image.

YoSafeguard, 2023-04-21:
Got it!
Strange situation! Access to the receiver via FTP occurs by domain name. The receiver itself works through a VPN. From two computers in the local network it comes in normally (namely, by domain name), but from the third (from the outside) it does not want to! I turn off the VPN on the receiver and the third one also starts to connect normally. 
But - this is my problem! 
Thank you!

DYefremov, 2023-04-21:
Ok. We can close the topic.

closed 2023-04-21 by DYefremov