DYefremov/DemonEditor #178: EXE  file

ruijmnunes, 2023-05-24:
Where is exe file?

Thanks

DYefremov, 2023-05-24:
Hello,
First of all, I recommend reading carefully  the '[Installation and Launch](https://github.com/DYefremov/DemonEditor#installation-and-launch)' section of the [README](https://github.com/DYefremov/DemonEditor#readme).

Latest build for MS Windows can be downloaded [here](https://github.com/DYefremov/DemonEditor/releases/download/3.6.2-b1/DemonEditor-3.6.2-Beta-win64-MPV.zip).

ruijmnunes, 2023-05-24:
I try download the files and change with DenomEditor in program files folder, now the program dont start.

Any idea?

DYefremov, 2023-05-24:
Unfortunately without detailed information about where (which OS) how and what you launch, its can be only guess. You can run the program using the _DemonEditor-dbg.exe_ file from the root folder. The program should start with a terminal window in which you can see possible errors.

ruijmnunes, 2023-05-24:
[image]

DYefremov, 2023-05-24:
Hm, very strange. It looks for icons and not findes. What version of Windows and monitor resolution do you have? Delete the program from the 'Program Files' folder and unpack it to another location. Also try the [previous version](https://github.com/DYefremov/DemonEditor/releases/download/3.6.1-b1/DemonEditor-3.6.1-Beta-win64-MPV.zip) or [this installer](https://github.com/DYefremov/DemonEditor/releases/download/3.6.0-b1/DemonEditor-3.6.0-Beta-x64-Install.exe).

ruijmnunes, 2023-05-24:
I use windows 11

with 3.6.0 Beta,have exe to install work great, but with this files no.

DYefremov, 2023-05-24:
In the latest versions have been updated libraries set. Perhaps this is the reason. Nevertheless, everything works for me on Windows 11. Please [test this](https://github.com/DYefremov/DemonEditor/releases/download/3.6.0-b1/DemonEditor-3.6.0-Beta-win64-MPV.zip) too!

ruijmnunes, 2023-05-24:
work

DYefremov, 2023-05-25:
> work

Doubly strange. Apparently there really is a problem with one of the libraries. I'll try to update again in the next build. 
Unfortunately, I can't reproduce this problem at me. Everything starts and works as it should.

DYefremov, 2023-06-04:
New [build](https://github.com/DYefremov/DemonEditor/releases/download/3.7.0-a1/DemonEditor-3.7.0-Alpha-win64-MPV.zip) available. Please test!

ruijmnunes, 2023-06-04:
Hello!

Dont start, same problem.

only DemonEditor-3.6.0-Beta-x64-Install work good!

Thanks

DYefremov, 2023-06-04:
Hmm, tested on 2 different Windows 11 PCs. Everything runs. It cannot be ruled out that the question is about Windows security. There was a report that the antivirus "cursed" on executable files. So far no ideas.

DYefremov, 2024-01-06:
Because it was long time no more messages with a similar problem, we are closing it for now.

closed 2024-01-06 by DYefremov