DYefremov/DemonEditor #182: error transfering boquets to receiver

snickers2k, 2023-07-20:
Downloading Boquets from the DM works
```
20-07-23 04:34:03 Downloading file: blacklist.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: bouquets.radio.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: bouquets.tv.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: lamedb.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: userbouquet.favourites.radio.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: userbouquet.favourites.tv.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: whitelist.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: satellites.xml.   Status: 226 Transfer complete.
20-07-23 04:34:03 Downloading file: terrestrial.xml.   Status: 226 Transfer complete.
```

but uploading gives this erroor
```
20-07-23 04:34:03 *** Done. ***
20-07-23 04:34:33 Uploading data error: encoding with 'idna' codec failed (UnicodeError: label empty or too long)
20-07-23 04:34:33 ##### Done! #####
```

DYefremov, 2023-07-20:
Hi,
Unfortunately I can't reproduce it at me. But... Seems [like a problem](https://stackoverflow.com/questions/51901399/python-requests-encoding-with-idna-codec-failed-unicodeerror-label-empty-o) in one of the Python modules.
First of all, try to disable this option:
[image]
Also try to play with the value in the receiver address field. E.g, try setting as hostname.

DYefremov, 2023-07-20:
Hm...
I think it is impossible to rule out problems with your image (settings). You can upload the configuration files (for testing purposes) using a third party ftp client (or there is an FTP tab in the DemonEditor itself).

> i did ftp upload the boquets already, but can't find a way to import in the dreambox

I think here you need to refer to the documentation for your OS, image (perhaps on the support forum for this image). Perhaps there are some features.

Considering this:

> after disabling that, it does not give an error anymore, but instead crashes the dreambox

Perhaps it will be enough to correctly configure HTTP access in the Webinterface settings. For example disable "**token-based security**" and "**anti-hijack**". And turn back the HTTP option in DemonEditor. Also in recent versions there have been minor changes in data loading. Therefore, try [earlier versions](https://github.com/DYefremov/DemonEditor/releases) of the program.

closed 2023-11-08 by DYefremov