DYefremov/DemonEditor #185: Importing bouquet from the receiver using the e2m3u2bouquet plugin

YoSafeguard, 2023-10-10:
Hello!
To create and update bouquets of IPTV channels, I use the wonderful e2m3ubouquet plugin.
Recently, the author of the plugin updated the principle of generating channel references.
In addition, I also changed the references of channel bouquets in the bouquets.tv file
As a result, DemonEditor produces the following error when importing data from the receiver:

`10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:9d1f:FROM BOUQUET "e2m3u2b_tv-team_novye-tansliatsii.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:830e:FROM BOUQUET "e2m3u2b_tv-team_izbrannoe.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:0b7a:FROM BOUQUET "e2m3u2b_tv-team_kino.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:a159:FROM BOUQUET "e2m3u2b_tv-team_kinozaly.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:7f55:FROM BOUQUET "e2m3u2b_tv-team_federalnye.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:7d1b:FROM BOUQUET "e2m3u2b_tv-team_novosti.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:5772:FROM BOUQUET "e2m3u2b_tv-team_obrazovanie.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:dc02:FROM BOUQUET "e2m3u2b_tv-team_priroda.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:2fdd:FROM BOUQUET "e2m3u2b_tv-team_muzyka.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:8c18:FROM BOUQUET "e2m3u2b_tv-team_sport.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:4b76:FROM BOUQUET "e2m3u2b_tv-team_detskie.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:ccae:FROM BOUQUET "e2m3u2b_tv-team_prochee.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:4e2f:FROM BOUQUET "e2m3u2b_tv-team_dlia-vzroslykh.iptv" ORDER BY bouquet
].
10-10-23 11:49:51 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:dddd0000:5039:FROM BOUQUET "e2m3u2b_tv-team_uhd.iptv" ORDER BY bouquet
].`

Perhaps due to an extra character in dddd0000?
Please comment! Thank you!

DYefremov, 2023-10-10:
Hi,
In your case the problem is in the bouquet file name.E.g. `"e2m3u2b_tv-team_uhd.iptv"`.
Typically, a user bouquet file has the extension (suffix) _*.tv_ or _*.radio_. And also the prefix _userbouquet._*.
Therefore the bouquet file name should be: `"userbouquet.e2m3u2b_tv-team_uhd.tv"`
Perhaps this is not a standard (and your file names are valid), but all the settings (as all descriptions and rules about bouquets creation) that came to me , as well as the bouquets created by Enigma2 itself, adhere to this format.

YoSafeguard, 2023-10-11:
Hello!
Thanks For The Clarification. I Will Write To The Plugin Developer.

YoSafeguard, 2023-10-11:
Прошу прощения, можно по-русски?
Вот что мне ответил автор плагина:

Ответ преолестен ))) особенно умиляет вот это

Typically, A User Bouquet File Has The Extension (Suffix) *.Tv Or *.Radio. And Also The Prefix Userbouquet.*.

Это уровень! Передайте разработчику что ДО ЛАМПОЧКИ какие имена имеют файлы "пользовательских" букетов ... Важно как сформирован КОРНЕВОЙ файл bouquets.tv и от этого "плясать" ... Если разработчик с этим НЕ согласен, предлагаю задуматься над простым вопросом: "почему ЛЮБОЙ имидж E2 прекрасно переваривает имена файлов пользовательских букетов *.iptv  и вы видите букеты в фаворитных, осуществляете навигацию по ним , выбираете каналы и, о БОЖЕ у вас все работает?" ... Что за нах! Ведь исходя из логики ответа разработчика - нихрена не должно работать !?
Прежде чем утверждать вот это

herefore The Bouquet File Name Should Be: "Userbouquet.E2M3U2B_Tv-Team_Uhd.Tv"
Perhaps This Is Not A Standard (And Your File Names Are Valid), But All The Settings (As All Descriptions And Rules About Bouquets Creation) That Came To Me , As Well As The Bouquets Created By Enigma2 Itself, Adhere To This Format.

разработчик MUST AND SHOULD тщательно изучить стандарт ETSI EN 300 468 реализацией которого и есть framework Enigma2 где напрочь отсутсвуют озвученные им требования к пользовательским букетам, а все его "утверждения" строятся на том что кто-то в коде питон имиджа придумал и захардкодил что при создании "фаворитного" букета с пульта его имя автоматически получает префикс "userbouquet" .... с какого перепугу это вдруг стало стандартом ?

В общем я могу привести "разработчику" пример из 10 строк кода на питоне который будет на основании корневого файла bouquets.tv формировать список файлов букетов ВНЕ зависимости от "префиксов" и "суффиксов" да еще и проверять валидность этих записей ;D  , но надеюсь он парень талантливый и таки изучит ETSI EN 300 468 и исходники ЛЮБОГО имиджа E2 где сам увидит этот код на питоне и осознает "легкую несуразность" своего ответа 

p.s. Кстати логика программера писавшего код на питоне функционал создания пользовательских букетов - мне абсолютно понятна ....  весьма логично что если ПОЛЬЗОВАТЕЛЬ с пульта создает свой ПОЛЬЗОВАТЕЛЬСКИЙ букет, то я бы тоже "влупил" бы ему префикс "userbouquet" ... как собственно и суффиксы "намекающее" к чему этот букет имеет отношение "tv" или "radio" ... но с какого перепугу логика программера стала стандартом и MUST распространятся на все и вся букетов - загадка космического масштаба  ;D

Может быть получится рассмотреть вопрос об отмене проверки названий файлов букетов?

DYefremov, 2023-10-11:
Hi,
I will answer in Russian also.
Думаю автор плагина слегка торопиться делать выводы и давать оценки другим :)

> ДО ЛАМПОЧКИ какие имена имеют файлы "пользовательских" букетов

Это не СОВСЕМ так! Далеко не всем компонентам образов с Enigma2 может это понравиться. Например некоторым библиотекам или сервисам имеющие дело с файловой системой (они могут и не знать про ETSI EN 300 468). Строго говоря Enigma2 это просто фронтенд (GUI) запущеный на Linux (если совсем упрощать).  То, что большинство свежих образов с Enigma2 может "скушать" почти все что угодно,вовсе не означает что так же будет и на более старых и уже не поддерживаемых версиях. А таких пока еще не мало. 

Касаемо "суффиксов и префиксов". Если вы пишите расширение для какого либо инструмента работающего с данным, логично воспользоваться принятыми методами и способами данного инструмента. Например вы пишите графическую оболочку для zip-архиватора. Думаю логично использовать расширение *.zip а не *.arch и.т д. Кроме того расширение файла могут использовать другие службы и сервисы ОС. Примерно из этой логики исхожу и я. Ниже пару ссылок на компонент Enigma2 который использует схожую логику и которая используется в DemonEditor хоть и в несколько ином виде. 
1.[getBouquetsList](https://github.com/OpenPLi/enigma2/blob/b269170e65447dc0cd74e019175d1a0cafd53bfb/lib/python/Components/ImportChannels.py#L84)
2.[ImportGetFilelist](https://github.com/OpenVisionE2/enigma2-openvision/blob/d274e308634ddcd8f4e40b9b70a003783d99f9e0/lib/python/Components/ChannelsImporter.py#L186C49-L186C159).

Теперь по **существу** вопроса. 
 Да, согласен. Как любой архиватор может **открыть** или попытаться открыть правильный файл архива (не зависимо от того есть у него нужное расширение или нет)  логично сделать это и в нашем случае при открытии списка букетов. 

> Может быть получится рассмотреть вопрос об отмене проверки названий файлов букетов?

Уже думал об этом. Просто есть участки кода, где используется тип букета. По возможности посмотрю, что можно сделать. Как минимум в режиме чтения.

YoSafeguard, 2023-10-12:
Приветствую!

> Думаю автор плагина слегка торопиться делать выводы и давать оценки другим :)

Это у него манера общения такая 😀 На форуме уже все привыкли

> Уже думал об этом. Просто есть участки кода, где используется тип букета. По возможности посмотрю, что можно сделать. Как минимум в режиме чтения.

Спасибо за развернутый комментарий! Буду ждать возможного решения, как минимум в режиме чтения 😀

DYefremov, 2023-10-16:
Убрал проверку имени файлов букетов при чтении. Особо не тестировал!  Измеменения доступны в ветках [development-3.0](https://github.com/DYefremov/DemonEditor/tree/development-3.0) и [testing](https://github.com/DYefremov/DemonEditor/tree/testing).

YoSafeguard, 2023-10-19:
Спасибо большое! Протестирую и отпишусь!

DYefremov, 2023-11-06:
[Test builds.](https://github.com/DYefremov/DemonEditor/releases/tag/3.8.0-test)

closed 2024-01-06 by DYefremov