DYefremov/DemonEditor #187: deleting working stream

luxorko, 2023-11-04:
testing demon 3.8.0alph;  Win10

IPTV - remove all unavaiable = lost working streams are deleteng 
could you check please 

`https://cp1.server89.com:19360/dasberni/dasberni.m3u8`

[image]

DYefremov, 2023-11-04:
I don't see any problems with this stream. Doesn't delete anything for me.

luxorko, 2023-11-04:
he does

please use same ref. number 

`4097_0_1_3604_3565_650d_1010101_0_0_0`

if you have ref number basic - `4097_0_1_0_0_0_0_0_0_0`

channel are not deleting...

DYefremov, 2023-11-04:
> he does

At me doesn't.

[image]
[image]

luxorko, 2023-11-04:
strange, maybe becouse im WIN10?

DYefremov, 2023-11-04:
On screenshots Windows 10 also.

luxorko, 2023-11-04:
what could be difference?
any ide?

i see you did some update on demon
my wersion are from 06 sep ..do you have same or new one?

DYefremov, 2023-11-04:
I tested with [this version](https://github.com/DYefremov/DemonEditor/releases/tag/3.8.0-a1).

luxorko, 2023-11-04:
im using same...

luxorko, 2023-11-04:
Could yu check my bouquet, plese 
...all streams will deleting....
"IPTV- Remove all unavaibe"

[
[userbouquet.e2m3u2b_italia-tuvi_new-broadcasts.tv.txt](https://github.com/DYefremov/DemonEditor/files/13257585/userbouquet.e2m3u2b_italia-tuvi_new-broadcasts.tv.txt)
](url)

DYefremov, 2023-11-04:
> im using same...

Depending your screenshot this is an older version. Show your About dialog!

luxorko, 2023-11-04:
[image]

luxorko, 2023-11-04:
> Could yu check my bouquet, plese ...all streams will deleting.... "IPTV- Remove all unavaibe"
> 
> [ [userbouquet.e2m3u2b_italia-tuvi_new-broadcasts.tv.txt](https://github.com/DYefremov/DemonEditor/files/13257585/userbouquet.e2m3u2b_italia-tuvi_new-broadcasts.tv.txt) ](url)

just remove ".txt "

DYefremov, 2023-11-04:
Download again and run from the unpacked folder. Perhaps you unpacked over an old version or something similar.

luxorko, 2023-11-04:
done. but same story.... streams deleted

[image]

luxorko, 2023-11-04:
fill list/bouquets
[image]

after "remove all unavaiable" stsays just marker
[image]

luxorko, 2023-11-04:
did you test sending bouquet?

DYefremov, 2023-11-04:
Yes. I’m not completely sure of the correctness of this file, but I see one potential problem. Perhaps this is really a bug. I'll take a closer look later.

luxorko, 2023-11-04:
> I'll take a closer look later.

hope you will find and fix bug and new version will be relesed
thank you for hard work

luxorko, 2023-11-05:
thank you, believe is working
could you relesed new version for Win10 please

DYefremov, 2023-11-06:
[Test builds.](https://github.com/DYefremov/DemonEditor/releases/tag/3.8.0-test)

closed 2023-11-08 by DYefremov