DYefremov/DemonEditor #189: EPG link not work

marcus83ita, 2023-11-13:
hi,

When I insert the xml links I always get this error, are there any solutions? Thank you!

https://www.bevy.be/bevyfiles/italypremium.xml

[image]

DYefremov, 2023-11-13:
Hi,
Firstly link directly on _*.xml_ file will not work. By default, the file must be compressed ("_.gz", ".xz", ".lzma"_ are supported). Of course, we can add support for uncompressed _*.xml_. There is no big problem.

As for the SSL error, there may be different variants. It depends from OS. What OS (version) are you using? And also the version (build) of the program.

marcus83ita, 2023-11-13:
yes,

I also tried with *.gz, (http://www.teleguide.info/download/new3/xmltv.xml.gz) it tells me file not supported...see the screenshot:

[image]

I use macOS Big Sur

marcus83ita, 2023-11-13:
also *.xz:

 file not supported...

marcus83ita, 2023-11-13:
I also tried with Windows 11, same problem here too

DYefremov, 2023-11-13:
This feature still very experimental, but... I made quick test:
[image]
Unfortunately I can only guess. It may be an issue with your internet connection. E.g., you are using a VPN.

DYefremov, 2023-11-13:
This feature still very experimental, but.. I made quick test:
[image]

Unfortunately I can only guess. There might be something wrong with your internet connection. E.g., you are using a VPN.

marcus83ita, 2023-11-13:
I say IPTV channels. can you try there? i don't use a VPN

DYefremov, 2023-11-13:
Depending your screenshots, you have a problem with file download, and not with displaying EPGs. It doesn't matter what channel type.
[image]

marcus83ita, 2023-11-13:
ok thanks man I follow that post, thanks anyway for the support! have a I wish you a good day/evening!

marcus83ita, 2023-11-14:
hi! in the next update add the uncompressed xml EPG link, it would be better because there are more sources. thx!

marcus83ita, 2023-11-14:
I tried to load the EPGs with DemonEditor but it doesn't load them... I transfer to the receiver but it doesn't load them, I also removed EPGimport the same thing, Is there any step to send EPGs with the software?

marcus83ita, 2023-11-14:
non capisco perchè mi da errore nei canali EPG, while in satellite channels the link loads it normally...

[image]

DYefremov, 2023-11-14:
Hi,
Unfortunately, you apparently did not read my previous answer carefully. This tool used to assign references to channels, not for downloading EPG. The format of the *xml file there is completely different and serves other purposes. I recommend using the [search](https://github.com/search?q=repo%3ADYefremov%2FDemonEditor%20EPG%20assign&type=issues). I think it will become clearer.

> Is there any step to send EPGs with the software?

No. The current function discussed above serves to test EPG sources before using them for example in the EPGimport plugin. Useful for finding the best fit for your bouquets. In addition, some users use this program to watch TV on a PC. And EPG display sometimes is very useful. I was thinking about a separate extension for updating the EPG in the receiver in the form of a [separate extension](https://github.com/DYefremov/demoneditor-extensions) for this program, but there is no time for this yet.

> in the next update add the uncompressed xml EPG link

Ok. I'll add it either in this or the next version.

marcus83ita, 2023-11-14:
> No. The current function discussed above serves to test EPG sources before using them for example in the EPGimport plugin. Useful for finding the best fit for your bouquets. In addition, some users [...]

now I understand perfectly, I hope in the future the option to directly load the EPGs from the software will be integrated, thanks again my friend!!

DYefremov, 2024-02-15:
FYI. As promised, I made a small [extension](https://github.com/DYefremov/demoneditor-extensions/tree/main/extensions/epgexport) for exporting EPG from XMLTV. If necessary, can be [discussed here](https://github.com/DYefremov/demoneditor-extensions/discussions/3).

marcus83ita, 2024-02-15:
Was this extension put in version 3.0.9? i use macOS

[image]

DYefremov, 2024-02-15:
Hi,

> use macOS

As described [here](https://github.com/DYefremov/demoneditor-extensions/tree/main/extensions/epgexport#readme) and also seen from your screenshot, this extension requires program version 3.9.0 and higher. For macOS users (if you are using a ready-made build), the Alpha version will be soon available. Just stay tuned to the releases page for updates. If time permits, I will post it in the coming days.

marcus83ita, 2024-02-15:
thanks bro! I'm waiting for the macOS version

luxorko, 2024-02-16:
> Alpha version will be soon available.

could you be able make version for Windows, please
Thank you

marcus83ita, 2024-02-16:
> > Alpha version will be soon available.
> 
> could you be able make version for Windows, please Thank you

it is already available for windows >>> https://github.com/DYefremov/DemonEditor/releases

marcus83ita, 2024-02-18:
it loads the xmltv file perfectly to the box, but why doesn't it recognize the IPTV ones? look the screenshot

[image]

marcus83ita, 2024-02-18:
but now it is no longer possible to import m3u lists?

[image]

DYefremov, 2024-02-18:
> but now it is no longer possible to import m3u lists?

Import/export elements have been moved to the bouquet list panel and context menu.
[image]

Regarding this question:

> it loads the xmltv file perfectly to the box, but why doesn't it recognize the IPTV ones?

Unfortunately, you apparently still don’t understand what this dialog/tool is used for. :(
Ok...
This dialog serves for convenient assignment of satellite channel references (_sid, nid, tid, namespace_ values) to IPTV channels of the selected bouquet. 
On the left, you load a list of current satellite channels from which you, by dragging (or copy/past) to the selected IPTV service, assign the value of the satellite channel reference to the channel(s) to which you dragged (pasted). Additionally, there are features  that simplify and speed up this process. For example, filtering and auto-selection. Additionally, it is possible to show only the list of channels for which there is a program in the EPG cache (epd.dat file). Why do you need the entire list, for example, of 5 thousand channels, if you don’t have a program for them. t is this function that you activated, according to your screenshot. By assigning links in this way, you don’t have to worry about obtaining a program for your IPTV services  using additional tools, but use the one received from the satellite(s).

If you want to use third-party _XMLTV_ sources and create an EPG cache yourself, then you can use additional tools. E.g. The _**EPG Import** plugin_.  In addition, if you want to use this program and do not want to use plugins for Enigma2, then from version **3.9.0** it will be possible to create and load a cache file (`epg.dat`) from this program via additional [extension](https://github.com/DYefremov/demoneditor-extensions/tree/main/extensions/epgexport). 
[Here you can find Quick HowTo](https://github.com/DYefremov/demoneditor-extensions/discussions/3) and discuss possible problems or suggestions.

marcus83ita, 2024-02-18:
yes, sorry, I didn't understand completely... now yes, but I'm looking for how to load EPGs into IPTV channels, but it seems impossible :(

DYefremov, 2024-02-18:
> but it seems impossible

Not at all. I**t's possible!**
If you use free sources of playlists and EPGs, then very often you need to "make friends" these 2 things.

The most common problem that I have encountered is that the **names** (IDs) of the services in your playlist and the *.xml file of the EPG source do not match.
I did a small test for 1 channel (Rai 1) for this sources:
* [EPG source.](https://www.bevy.be/bevyfiles/italypremium.xml)
* [M3U source.](https://iptv-org.github.io/iptv/countries/it.m3u)

[image]

[image]

In my case, I needed just rename channel name correctly in the favorites list. Perhaps it makes sense to think about an additional tool that allows at least partially automate this process.

marcus83ita, 2024-02-19:
you have to rename the channels manually, it's too stressful, I hope you will be able to implement this function automatically in the future. thanks friend!

marcus83ita, 2024-02-19:
I can't find the TAB Log...🤔

[image]

DYefremov, 2024-02-19:
The log panel appears automatically upon certain actions. E.g sending and receiving data. You can also enable it manually via menu -> Tools -> Logs

DYefremov, 2024-03-07:
Made small discussion cleanup.