DYefremov/DemonEditor #192: [3.9.0-Alpha-win64] Unsupported or invalid data format

Heart1010, 2024-02-17:
Got these "errors" while firt loaded the data from my receiver (SF8008, pure2 v7.3)

```
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Cinema_&_Serien_-_high.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Cinema_&_Serien_-_max.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_SPORT_Deutschland.tv.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Dazn_Events_&_NFL.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_TR_Bein_&_EXXEN_Sports.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_RTL+_(live_Events).tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_PDC_Board_(Darts).tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_ES_Cine&Serie.tv" ORDER BY bouquet
].
17-02-24 13:51:22 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_NL&BE.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Cinema_&_Serien_-_high.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Cinema_&_Serien_-_max.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_SPORT_Deutschland.tv.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_Dazn_Events_&_NFL.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_TR_Bein_&_EXXEN_Sports.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_RTL+_(live_Events).tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_PDC_Board_(Darts).tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_ES_Cine&Serie.tv" ORDER BY bouquet
].
17-02-24 13:51:23 Unsupported or invalid data format: [#SERVICE 1:7:1:0:0:0:0:0:0:0:FROM BOUQUET "userbouquet.bouquetmakerxtream_live_iptv_NL&BE.tv" ORDER BY bouquet
].
```

DYefremov, 2024-02-17:
Hi
Such symbols are not accepted. How these settings were obtained? I can add this symbols to exclusion but not sure.

Heart1010, 2024-02-17:
The plugin "bouquetmakerxtream" did make these names.... and no problem with enigma2/pure2 to use them until now.

DYefremov, 2024-02-17:
> and no problem with enigma2/pure2 to use them until now

Such symbols are 'prohibited' for a reason! After all, there is always a desire to 'share' your settings with others. And they may run into problems. Especially on older images. Here needs to think. Perhaps it makes sense to disable the check and give the user complete carte blanche. :)

DYefremov, 2024-02-17:
Quickly made this option:
[image]
Added addition checking and warning to log. Didn't test it much.

Heart1010, 2024-02-21:
So I get the warning to the log but can use these type of userbouqet file name(s) in one of the next versions? 👍

DYefremov, 2024-02-21:
Yes,  it will be displayed a warning , but the bouquets must be loaded. But as I wrote above

> Didn't test it much.

closed 2024-03-06 by DYefremov