DYefremov/DemonEditor #194: userbouquet

navidyazdan, 2024-02-25:
Some service in userbouquet doesn't load in app

DYefremov, 2024-02-25:
Sorry, but due to the complete missing of info (what bouquet, what services) it is difficult to comment on anything. It would be optimal if you send me a bouquet file indicating the services you are having a problem with.

navidyazdan, 2024-02-25:
[image]
Line 3 and Line 9

DYefremov, 2024-02-25:
it looks ok. Please post (as attachment) or send me your lamedb(5) file.

DYefremov, 2024-02-29:
I had a little time to experiment...
Based on the above screenshot and service parameters, this should be a satellite in position 52.0E. Having checked several sites, for example [LyngSat](https://www.lyngsat.com/TurkmenAlem-MonacoSat.html) and others, I did not find any services with similar parameters. Perhaps you could indicate the names of the services? In addition, having downloaded several public sets of settings for motors (Vhannibal and the like), no such services were found also. Perhaps I missed something...

In general, without a set of your settings, it is difficult for me to confirm and check your case.

navidyazdan, 2024-03-01:
[](url)
[image]
This is  bouquet from my Zero4k Gem Series HD and River HD is in bouquet

navidyazdan, 2024-03-01:
[image]

This picture from demon editor there is no Gem Series HD and River HD

navidyazdan, 2024-03-01:
[image]
Total channel in Zero4K bouquet is 90

navidyazdan, 2024-03-01:
[image]
total channel in demon editor in 85

navidyazdan, 2024-03-01:
[lamedb5.zip](https://github.com/DYefremov/DemonEditor/files/14464220/lamedb5.zip)

my lamedb(5) file

navidyazdan, 2024-03-01:
I have not this problem with dreambox edit only with demon editor

DYefremov, 2024-03-01:
Yes, this is an bug in reading _lamedb5_. Support for this format is still experimental.  I've added small fix.

[image]

If possible, check your settings for ver. 4 also. Thx.

DYefremov, 2024-03-02:
I see you are using Windows. In your case, the easiest way is to wait for the update. Or, if you have the desire and time, you can try to [build and run](https://github.com/DYefremov/DemonEditor/blob/master/build/BUILD_WIN.md) it yourself. :)

navidyazdan, 2024-03-05:
Thanks Dmitriy, for your help :-*

DYefremov, 2024-03-06:
With the hope that everything is working we can close.