DYefremov/DemonEditor #196: Question about download-upload settings

risky777, 2024-02-28:
Download and save (to file) depends on active tab, and I don't know how it behaves on upload. Is this bug? Can you explain what is the procedure?

Explanation:
When in tab "Bouquets", bouquets and satellites.xml are downloaded from enigma2, save clears folder and saves only bouquets, so satellites.xml is not there. Because of that, I have to go to tab "Satellites" and download from enigma2 (this downloads only satellites.xml and similar) and use save on this tab, now satellites.xml is saved.
When uploading I do this procedure in reverse ("Satellites" and then "Bouquets").

DYefremov, 2024-03-06:
With the hope that everything is working we can close.

risky777, 2024-03-16:
Tried version 3.9.1-beta, and nothing changed. Save on Bouquets tab still doesn't save satellites.xml.

DYefremov, 2024-03-16:
Just try changes from [development](https://github.com/DYefremov/DemonEditor/tree/development-3.0) branch.