DYefremov/DemonEditor #200: How possible would it be to build a lyngsat scraper for channel names?

ilikenwf, 2024-05-05:
Since we should already have all the data on a tp and satellite couldn't we just scrape lyngsat to autopopulate the channel names?

ilikenwf, 2024-05-05:
I mean one that fully functions - is this a regex thing?.

[image]

DYefremov, 2024-05-06:
Hi,
> is this a regex thing?.

Yes, it is. You can experiment [with it](https://github.com/DYefremov/DemonEditor/blob/8330104f3ca53fdfd9fb0f7f082b1a79eb366ad7/app/tools/satellites.py#L446) itself. When I have the opportunity, I'll also take a look.

DYefremov, 2024-05-06:
I have checked [this satellite](https://www.lyngsat.com/Galaxy-37-Horizons-4.html). For most transponders there missing **ONID** and **TID** values.

ilikenwf, 2024-05-06:
...onid may be missing, per se, but because the logos for each channel are there could we not determine the network for EPG that way?

Even just scraping the channel names would be beneficial as many of them are named "Program X" on the TPs of this bird.

ilikenwf, 2024-05-06:
On the muxes page the picons are named (inconsistently) but do have the network name in the file name.

DYefremov, 2024-05-06:
In most cases, we need the TID and ONID values to correctly identify service (constructing service ID). The services may have identical name but different IDs (on different satellites-transponders). For some cases the TID and NID values can be '0'. I don't know how correct this, but I seen some times this variant. In your case it's possible to make making '0' as fallback values if no TID and NID values are not founded. But we need always think, that this may work not properly.

ilikenwf, 2024-05-06:
Since I'm thick headed, do you have a link for a mux that has that information populated so I can more easily see what it is that is missing?

I've personally scanned this transponder, so if I need to join lyngsat and submit updates I can try to do that...

It's really too bad satbeams and lyngsat are both so out of sync and not always up to date.

DYefremov, 2024-05-06:
> do you have a link for a mux that has that information populated

It is not fully clear from your message, what info you need.

ilikenwf, 2024-05-06:
Nevermind, haha.

I was out of this hobby for about 15 years, still getting acquainted with it. Please let me know if I can assist more than the regex, I'll try to look at it if and when I get a chance.

DYefremov, 2024-05-06:
Ok.I think we can close topic.

closed 2024-05-06 by DYefremov

ilikenwf, 2024-05-10:
> Hi,
> 
> > is this a regex thing?.
> 
> Yes, it is. You can experiment [with it](https://github.com/DYefremov/DemonEditor/blob/8330104f3ca53fdfd9fb0f7f082b1a79eb366ad7/app/tools/satellites.py#L446) itself. When I have the opportunity, I'll also take a look.

First one matches properly:

[image]

Second one is broken unless you remove "?" operator from the start of the regex:

(T2-MI)?\s?(PLS\s+Multistream)?\s? matches:

[image]

But each match just matches empty space...

The third has no matches.

@DYefremov what exactly do we need to match to get this right for matching?

The TID is per mux on lyngsat - each mux is a separate transponder, so we get the TID that way. Then we can use SID, VPID, APID to correlate the values in the lamedb,

ilikenwf, 2024-05-10:
Lamedb (which I'm sure you're already fully knowledgeable of since you built this tool):

https://www.satsupreme.com/showthread.php/194074-Lamedb-format-explained

ilikenwf, 2024-05-10:
We can capture the FEC value, but is it really needed?

Ultimately if we just want to get the channel names, per tp wouldn't we just need to match the pids up?

DYefremov, 2024-05-10:
Hi,
1. There is a feeling that you are testing on a limited number of satellites without comparing the real data received from the receiver with the data that is on the LyngSat website. Very often the data may differ or wrong.
2. Lamedb.  I have read enough info including this old link to the description that you provide. BUT more importantly, I have a large number of example configurations created by various receivers with various versions of Enigma2, and not by editors! And sometimes the data may formed really different. Some values may missing of include additional flags. Based on this, if something does not work correctly for you when receiving (scraping) data from the **Web**, then in most cases the problem is in the input data or the algorithm for obtaining it and not in generating the `lamedb`.

> FEC value, but is it really needed?

Yes, at least the value 'auto'.

> Ultimately if we just want to get the channel names

If your task is to obtain channel names, then this is not what this program (scrapping feature) is intended for. One satellite and even one transponder may have several channels with the same names. The function under discussion first generates channel ids (references) and then places them in the lamedb including, of course, channel names. 

If you have **practical suggestions** for changing the code to improve scraping, I will be only glad. A pull request are welcome!

ilikenwf, 2024-05-10:
1. I'm testing on 127W and 125W, but the way in which Lyngsat presents data is standardized. Likewise I know my data from my receiver is correct, doing multiple scans and comparing them against the lyngsat and satbeams entries.

2. I figured this to be the case, however in our case the issue with lyngsat is that they aren't presenting the data the same way as kingofsat, and as such, we can't fetch data from them properly.

I do want channel names but your tool fulfills this purpose and could also prepopulate my database for other satellites as I scan the arc. I realize that there are cases where channels double up. 

Generally speaking I know regex is more efficient for scraping data but using beautifulsoup could reduce the pain - lyngsat is poorly constructed html as they use tables for EVERYTHING like it is 1992.