DYefremov/DemonEditor #201: lamedb services duplication

risky777, 2024-06-05:
It seems that demoneditor duplicates services in lamedb.

I've checked with ftp directly without demoneditor, lamedb on my enigma2-box (openatv-6.4) is always ok, without duplicated services, even when (faulty) lamedb from demoneditor is uploaded to enigma2-box.

But, when demoneditor downloads lamedb it duplicates services which is visible in demoneditor services-list and in lamedb itself.

I've noticed this on termux-android-arm64 and then tested on archlinux-amd64, demoneditor-3.10.

DYefremov, 2024-06-05:
The editor is designed in such a way that already during the loading services from lamedb (at least for **version _4_**) it will prevent the presence of duplicates! If you are sure that such a bug exists, then I need at least a lamedb file **before loading** it into the editor, and then **after saving** the data in the editor, to check your hypothesis!

DYefremov, 2024-06-06:
Yes, the first file contains duplicates. But...
I have made short tests in _macOS-arm64_ and _Manjaro Linux-amd64_. If I load a file with duplicates into the program, the duplicates are not displayed (as they should be), and after saving they disappear from the file.

[image]

I tested it for another file also. Results after saving in attachment.

[TEST.zip](https://github.com/user-attachments/files/15626045/TEST.zip)

At the moment I can't reproduce this problem at me. Perhaps the problem occurs under certain conditions... For now needs testing further.

risky777, 2024-06-07:
I'll try some older versions of demoneditor, because I don't recall happening it before.

risky777, 2024-06-08:
Version 3.8.0 doesn't have this bug, version 3.9.2 has it. Didn't had versions inbetween so still have to check those.

DYefremov, 2024-06-08:
Yes, I can confirm. A very interesting case. This can happen when loading from the home page when changing the profile to something other than the default on first data loading.

DYefremov, 2024-06-08:
Made small fix. Just for testing. Hope it helps. It may needs some other additional changes. 

P.S. It looks like the error occurred when implementing the auto data loading feature on desired tab selection from startup page.

risky777, 2024-06-15:
Fixed. Thanks.

Closed issue.

closed 2024-06-15 by DYefremov