DYefremov/DemonEditor #202: No Bouquet details after download

PBdeWit, 2024-06-16:
I'm using this program on a Mac Book Air with a M1 chipset

This is the log file after download
HTTP API initialized... Web Interface version: OWIF 1.3.9
HTTP API initialized... Web Interface version: OWIF 1.3.9
FTP OK.
Downloading file: blacklist.   Status: 226 Transfer complete.
Downloading file: bouquets.radio.   Status: 226 Transfer complete.
Downloading file: bouquets.tv.   Status: 226 Transfer complete.
Downloading file: lamedb.   Status: 226 Transfer complete.
Downloading file: lamedb5.   Status: 226 Transfer complete.
Downloading file: userbouquet.favourites.radio.   Status: 226 Transfer complete.
Downloading file: userbouquet.favourites.tv.   Status: 226 Transfer complete.
Downloading file: whitelist.   Status: 226 Transfer complete.
Downloading file: cables.xml.   Status: 226 Transfer complete.
Downloading file: satellites.xml.   Status: 226 Transfer complete.
Downloading file: terrestrial.xml.   Status: 226 Transfer complete.
*** Done. ***
Error. No matching service type [ServiceType  1] was found.
Unsupported or invalid data format: [#SERVICE: 1:7:1:0:0:0:0:0:0:0:userbouquet.favourites.tv

All list's are empty, Bouquets and also Bouquets details
[2024-06-16_13-44-20.zip](https://github.com/user-attachments/files/15857707/2024-06-16_13-44-20.zip)

Greetz Peter, wit.de.pb@gmail.com

DYefremov, 2024-06-16:
> 1:7:1:0:0:0:0:0:0:0:userbouquet.favourites.tv

How was the "_bouquets.tv_" file and favorites list created?  I could be wrong, but I have not seen yet formatting like in your file.

DYefremov, 2024-06-16:
I checked more closely...
Yes, this is a flaw.

closed 2024-06-21 by DYefremov