DYefremov/DemonEditor #203: Feature-request automatic "mark not present in bouquets"

risky777, 2024-06-21:
In services-list there is context-command "mark not present in bouquets". I made it automatic by following in "app/ui/main.py":

- add lines at end of function "fav_paste"
gen = self.mark_not_in_bouquets()
GLib.idle_add(lambda: next(gen, False), priority=GLib.PRIORITY_LOW)

- add lines at enb of function "remove_favs"
gen = self.mark_not_in_bouquets()
GLib.idle_add(lambda: next(gen, False), priority=GLib.PRIORITY_LOW)

I'm used to this behaviour from "dreamset" so I've been adding this for already several versions of demoneditor, and with them I've done bigger changes in bouquets, so it seems that it works ok. And, I do modifications on services-list filtered for only-one-satellite-at-time, so these additions are fast.

Maybe you could add this to official demoneditor, so I don't have to add it manually for every new version.

DYefremov, 2024-06-21:
> so I don't have to add it manually for every new version

That is unnecessary. Just use [extension](https://github.com/DYefremov/demoneditor-extensions). I did it quickly for your case.

[image]

[favautomarker.zip](https://github.com/user-attachments/files/15931284/favautomarker.zip)

Just unpack it into a folder:
`~/DemonEditor/tools/extensions`

or: 
`root folder of the program/extensions`

If everything works out, it can be added to the [repo](https://github.com/DYefremov/demoneditor-extensions/tree/main/extensions).

I haven't tested it completely! If you have questions or suggestions for improvement, it is better to open a separate issue [here](https://github.com/DYefremov/demoneditor-extensions/issues).

DYefremov, 2024-08-30:
The extension has been updated and added to the [repo](https://github.com/DYefremov/demoneditor-extensions/tree/main/extensions/favautomarker). Further discussion can be found [here](https://github.com/DYefremov/demoneditor-extensions/issues/4).

closed 2024-10-09 by DYefremov