DYefremov/DemonEditor #204: Feature request : seperate satellite names for C-band and KU-band

ahmedmoselhi, 2024-07-01:
Like in oe-allience sattellites.xml I want the web importer to have an option to make seperate sat names for both C & ku band i.e: 1.0°W Thor 5/6/7 becomes Thor 5/6/7 (KU) and Thor 5/6/7 (C) while using internet import service from lyngsat or kingofsat

DYefremov, 2024-07-02:
I'll look what can be done, but I can't promise a quick solution.

DYefremov, 2024-07-02:
I had a little time to experiment. :) I haven't tested it much.

ahmedmoselhi, 2024-07-02:
btw I'm on opensuse tumbleweed with kde 6.1 wayaland as main desktop

DYefremov, 2024-07-03:
Looks like I misunderstood you :) I made this option for importing satellites and not services! I'm not sure if this is a good option in this case, but we'll have to think...

DYefremov, 2024-07-07:
I have experimented a little more...
If you need names merging and separation by ranges in the satellites list at the startup, then I definitely won't do this option. This is due to the fact that transponder data is loaded only when necessary (lazy). Consequently, in order to divide satellites by band, it is necessary to download data for all transponders of all satellites, which is not good. f we are talking about combining names by sat position only, then this option is quite acceptable.

E.g so:
[image]

Оr, as an option, an additional button on the toolbar, by clicking on which you can combine satellites.

DYefremov, 2024-07-18:
Hmm, experimented with combining names in the sat list. In short, the disadvantages of implementing this feature for me are much greater than the advantages.:( E.g. Saving the last selected satellites, not to mention the need to seriously rework the logic when working with satellites. For now I'll leave it as it is. Perhaps there will be other options later.

closed 2026-01-19 by DYefremov