DYefremov/DemonEditor #208: Problem with channel names when uploaded to receiver

majster-pl, 2024-10-23:
After uploading to receiver some channel names are missing, image attached.
What I've noticed is that after not doing any changes I upload again and different channels will have names missing.
For testing I've used Dreambox for the same list and when uploaded all channel names appeared on receiver with no problems.

If you need any more details please let me know.

[image]

Logs after uploading:
[image]

> 23-10-24 22:23:24 ##### Uploading data on [10.4.4.152] #####
23-10-24 22:23:24 Testing HTTP connection...
23-10-24 22:23:25 HTTP: Sending info message...  Successful.
23-10-24 22:23:32 FTP OK.
23-10-24 22:23:32 Uploading file: satellites.xml.   Status: 226 Transfer complete.
23-10-24 22:23:32 Uploading file: terrestrial.xml.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: cables.xml.   Status: 226 Transfer complete.
23-10-24 22:23:33 Deleting file: userbouquet.dbe00.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe01.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe02.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe04.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe04.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe05.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe08.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe08.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe0a.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe0b.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe0c.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe10.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe11.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe12.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe20.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe21.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.dbe23.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.favourites.radio.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.favourites.tv.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Deleting file: userbouquet.favourites.tv.del.   Status: 250 Delete operation successful.
23-10-24 22:23:33 Uploading file: userbouquet.dbe08.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe08.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe00.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe04.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe0b.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.favourites.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe21.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe05.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: bouquets.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe0c.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe11.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe0a.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe01.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe20.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe04.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe10.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: bouquets.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe02.radio.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe23.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: userbouquet.dbe12.tv.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: lamedb.   Status: 226 Transfer complete.
23-10-24 22:23:33 Uploading file: blacklist.   Status: 226 Transfer complete.
23-10-24 22:23:34 HTTP: Reloading lamedb and Userbouquets. Successful.
23-10-24 22:23:34 ##### Uploading data on [127.0.0.1] #####
23-10-24 22:23:34 Testing HTTP connection...
23-10-24 22:23:34 HTTP test failed.
23-10-24 22:23:34 Uploading data error: [Errno 111] Connection refused
23-10-24 22:23:34 ##### Done! #####

Settings 1:
[image]

Settings 2:
[image]

DYefremov, 2024-10-24:
Hi, please send me your channels settings set. I'll check it at me. For now I can only guess that the problem may be in the encoding. How were the initial settings obtained?

majster-pl, 2024-10-24:
Hi @DYefremov thanks for coming back to me on this, here is a link to the list I was using: https://enigma2.hswg.pl/wp-content/uploads/2024/09/Lista-bzyk83-hb-13E-24.09.2024.zip
No sure what do you mean by initial settings, application initial settings? I think it was just default when first installed.
In terms of app version, I've also updated to 3.11.0 but actually now the list is populated with completely blank bouquets names and actually no channels in them... very odd.

[image]
[image]

DYefremov, 2024-10-25:
Hi,
Yes, I can confirm similar behavior on the latest _openATV ver 7.4_ image. But earlier ones working as should. After disabling the "Use HTTP to reload data in the receiver" option and reloading the data, it displays as expected. Yes, before sending the data, you need to click the save button! The program uses an approach of maximum preservation of previous, already loaded, data. Therefore, if you already have the settings loaded for the current profile, and you load new data via the "Open-Extract..." menu, then the previous data is not physically changed until you forcibly overwrite it!

[image]

It seems that the command for _WebIF_ with _**mode=2**_ (reloading bouquets and lamedb) does not work as required extent. At the moment, additional testing is needed to understand where is problem.

DYefremov, 2024-10-26:
Hi,
The problem turned out to be deeper than I expected...
Since this program was initially developed with an emphasis on the duplicating services possibility, then at the lamedb  loading  stage duplicates were filtered (not loaded). n the settings, that you provided, in lamedb, there are such entries (e.g. for the TVN HD service):
```
3dcd:00820000:0640:013e:1:0
TVN HD
```
```
3dcd:00820000:0640:013e:25:0
TVN HD
```
The difference is in the service type only! The program identifies such services as duplicates and when saving, the first of these services is not written to lamedb! In attach is a small mod of the program that takes this nuance into account. If possible, please test it!
[DemonEditor_3.11.1_mod.zip](https://github.com/user-attachments/files/17529625/DemonEditor_3.11.1_mod.zip)
[DemonEditor_3.11.1_Beta_MOD.deb.zip](https://github.com/user-attachments/files/17529633/DemonEditor_3.11.1_Beta_MOD.deb.zip)

majster-pl, 2024-10-26:
Morning Dmitriy! 
Yes I can confirm that your mod version works like harm :1st_place_medal: :rocket: 
Thank you for your time and commitment!
If you need me to test anything else please let me know.

[image]

DYefremov, 2024-10-26:
Excellent. Some more improvements and full testing of the program are required, as the changes affect most of the app. A little later I will upload the changes to git for general testing. 🚀

closed 2024-11-16 by DYefremov