DYefremov/DemonEditor #209: epg configuration - missing url

luxorko, 2024-11-28:
after adding references and click to "apply"
[image]

 all urls disappear - delete   using 3.11.2beta

DYefremov, 2024-11-28:
Hi,
Interesting case! Looks like changes in Python itself. :) 
Thx for the report. Should be fixed, but needs more testing.

DYefremov, 2024-11-29:
Hi, Please test [this](https://github.com/DYefremov/DemonEditor/releases/tag/3.11.3-b1) build!

luxorko, 2024-12-01:
> Hi, Please test [this](https://github.com/DYefremov/DemonEditor/releases/tag/3.11.3-b1) build!
hi confirm it is working properly now. thank you

I would like to ask if it will be possible to speed up the automatic selection?
now only works if channels name are 100% mach.

just found it on another forum, maybe it could help improve "automatically adding links to IPTV channels"

`https://www.linuxsat-support.com/thread/130052-plugin-e2m3u2bouquet/?postID=832758#post832758`

DYefremov, 2024-12-01:
> maybe it could help improve "automatically adding links to IPTV channels"

Yeah, I already thought about something like that. I think it would work for certain cases. Maybe it makes sense to make it optional. For example "Deep Search" etc. The only question is performance. Here needs to experiment.

At the moment we close this.

closed 2024-12-01 by DYefremov