DYefremov/DemonEditor #210: ffmpeg misdetection in Boot Logo tool

rdamas, 2024-12-01:
Version 3.11.3 Beta MacOS Arm Version installed from Releases dmg.

`ffmpeg` needs an additional lib that is not packaged in the dmg:

```/Applications/DemonEditor.app/Contents/Frameworks/ffmpeg 
dyld[35865]: Library not loaded: /opt/homebrew/opt/sdl2/lib/libSDL2-2.0.0.dylib
  Referenced from: <C86404EB-DFC9-31A1-8E95-3F036091FC12> /Applications/DemonEditor.app/Contents/Frameworks/ffmpeg
  Reason: tried: '/opt/homebrew/opt/sdl2/lib/libSDL2-2.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/sdl2/lib/libSDL2-2.0.0.dylib' (no such file), '/opt/homebrew/opt/sdl2/lib/libSDL2-2.0.0.dylib' (no such file)
Abort trap: 6```

This error is not detected because `ui.BootLogoManager.init` only checks for `FileNotFoundError`. I'd probably also check the output of `ffmpeg -version` for a plausible response.

DYefremov, 2024-12-01:
Yes, I can confirm this.
Apparently last PyInstaller version does not collect this library to the package. Needs additional testing.

DYefremov, 2024-12-01:
Replaced the binary from [here](https://osxexperts.net/) and updated the [ARM build](https://github.com/DYefremov/DemonEditor/releases/download/3.11.3-b1/DemonEditor-3.11.3-Beta-macOS-arm64.dmg). Please test.

rdamas, 2024-12-01:
That binary does the job.

closed 2024-12-01 by rdamas