DYefremov/DemonEditor #212: M3U IPTV

marcus83ita, 2024-12-13:
hi friend!

I just installed the latest version on MacOS, but there is no longer the possibility to load the m3u list...

Do I need to do a software downgrade? does this work will you put it in the future? thx!

[image]

marcus83ita, 2024-12-13:
I just found it, you have to right click on the name of the bouquet.

marcus83ita, 2024-12-13:
but I have a problem, it doesn't load the bouquets...sends them to the decoder, but does not load them🤔

DYefremov, 2024-12-13:
> I just found it, you have to right click on the name of the bouquet.

You can use these buttons also:

[image]

> but does not load them

Without logs and a clear understanding of what version of the program you are using, what playlist you are importing and how you are loading it is difficult to say anything. 
I briefly tested import from M3u in 2 versions (_3.10.2_ and _3.11.3_) and compared the resulting bouquet files. I didn't find any differences. Also, I loaded it into my receiver with openATV 7.4. Everything works. Version _3.11.*_ contains minor changes in refreshing data after downloading to the receiver. But in our case this should not affect, if I understand the problem correctly. In any case, I recommend trying earlier versions and checking.

marcus83ita, 2024-12-15:
I solved everything by loading with Windows software... in reality this problem has always appeared with MacOS

DYefremov, 2024-12-15:
Hi,
This is strange. I see no difference at me by data loading. I have tested with macOS  14.7 (_arm64_) also. Then needs a  log and sequence of actions of what and how was loaded. At least, I need a data set that you loading to test it at me. Maybe someone else will confirm a similar problem. At the moment I can't reproduce it at me.

marcus83ita, 2024-12-15:
at this point I think it's a Big Sur 11.7.10 problem...