DYefremov/DemonEditor #216: [Feature Request] Portable Mode

procobain, 2025-01-09:
Hello, I would like to be able to use DemonEditor in Portable mode. The reasons are varied:

Firstly, to ensure that data is not lost when formatting Windows.

Secondly, I use dualboot, I use Linux a lot more, and I would like to run DemonEditor with Wine (as I already do), but using the same data as a portable DE.

Cheers and thanks for the work.

DYefremov, 2025-01-09:
Hi,
If you read the description [(README](https://github.com/DYefremov/DemonEditor/blob/master/README.md)) carefully, it will be clear that the app was originally developed and works as a portable one. If I understood you correctly, then in your case it is enough to configure the paths for the data in the program settings. If we are talking about saving program settings itself (e.g. saving config file in the program folder itself), then I'll think about it.

procobain, 2025-01-10:
Yes, exactly what I was talking about, it's to save the configuration in a file in the folder itself, and regardless of where you run it, this file will be read...

I could run it on my W11, on my preferred prefix within Wine, on my notebook, and they would all have the same configuration, since I would be reading the configurations inside the file next to the folder.

DYefremov, 2025-01-10:
> on my preferred prefix within Wine

A very strange decision for me to launch a program originally developed for Linux. :)  As for the config file, as I wrote earlier, I'll think about it...