DYefremov/DemonEditor #217: [BUG] Demon Editor 3.11.3 Beta - Does not save changes to satellites.xml

procobain, 2025-01-20:
Does not save changes to satellites.xml

[image]

DYefremov, 2025-01-21:
Can confirm. Fixed.

procobain, 2025-01-21:
How to download the new version? The release dates are still old.

DYefremov, 2025-01-21:
If you want to test the latest changes on Linux you can run the program directly from the [sources](https://github.com/DYefremov/DemonEditor/archive/refs/heads/development-3.0.zip). In addition, it's possible to use [PPA](https://launchpad.net/~dmitriy-yefremov/+archive/ubuntu/demon-editor-testing), releases appear and tested there first. For other OS (e.g _Windows_), you can [build or run](https://github.com/DYefremov/DemonEditor/blob/development-3.0/build/BUILD_WIN.md) the app yourself or waiting for the build in [releases](https://github.com/DYefremov/DemonEditor/releases).

closed 2025-03-08 by DYefremov